thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #3: Set metrics programatically

alex-controlx, 2022-02-20:
Firstly, thank you for developing this valuable node!

A feature required in `mqtt sparkplug device` node: set metrics in  programatically or via **import** and **export** from a `csv` file.

I have around 500 metrics and setting them up via the UI is very tedious.

Programatically it can be by sending a payload with an array of names and types for metrics.

thomassorensen2650, 2022-02-28:
Hi Alex,

I think the easiest way would be to export the flow to JSON and edit the object there directly. 

I've added an undocumented and untested feature with no input checks whatsoever to version v1.1.0. It only works if you set the definition before birth is send. use with extreme causing, but you should be able to set metric definition like this:

```
msg.definition = {
    "MetricA" : "Int32",
    "MetricB" : "Int32"
}
```

I'll see if I can get it fully implemented in a future release.

alex-controlx, 2022-03-01:
Thank you for doing that. `msg.definition` will simplify my work.

I guess `dataType` property should have a value from the array at [https://github.com/thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus/blob/0543284ece8091372f22047b5f6e024bb0bb26b8/mqtt-sparkplug-plus.html#L355](https://github.com/thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus/blob/0543284ece8091372f22047b5f6e024bb0bb26b8/mqtt-sparkplug-plus.html#L355)

thomassorensen2650, 2022-03-01:
Correct, you can also send definition and payload in the same message.