thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #16: Primary host detection

artemz21, 2023-02-28:
Hi Thomas,
Could not find any solution elsewhere.

The Sparkplug B nodes work well except the "Store and Forward" feature, where the node should check the STATE message contents if the Primary Host application is online or offline.

When I enable this function (in this example the primary host name is "EMS"), the node keeps indicating "destination offline" even though I get the "online" STATE message for the connect topic /STATE/EMS (read by the regular MQTT node as the STATE message is not in the Sparkplug B format).

Ignition MQTT Engine is used as primary host application. 

Could you please help with this issue as the configuration seems to be done correctly at both ends. 
Thank you in advance.

[image]
[image]
[image]

thomassorensen2650, 2023-03-01:
From the screenshot it looks like you are sending state message in the spB1.0 namespace. That's not correct per the Sparkplug B specification, it should be send to this topic: /STATE/EMS

Another thing I noticed from the screenshot is that it looks like the payload is a javascript object, that's not the expected payload. It should be a text string "ONLINE" set to retained. Remember that you need to set the (non-retained) value to OFFLINE when primary host goes offline OR or on last-will. 

I do get quite a few questions about this feature, so I think it needs some better documentation. . I'll see if I can update the documentation or add an example to the node at some point.

artemz21, 2023-03-01:
Hi Thomas,
Thanks for the prompt response.
You are right, the STATE message I get from the Cirrus Link MQTT Engine is in the spBv1.0 namespace. However, the Sparkplug B specification 3.0.0 explicitly says it must be implemented this way. Please refer to section 5.13 of the spec (captured the screenshots below as well). The spec also says the payload must be in the JSON format.
[image]

I've just tested your implementation with /STATE/[Host] and just the string payload and it does work indeed. However, it doesn't look to be compliant with the 3.0.0 spec.

Looking forward to your feedback.

Cheers

thomassorensen2650, 2023-03-01:
Interesting! I did not know that they changed that in 3.0.0.

[Version 2](https://www.eclipse.org/tahu/spec/Sparkplug%20Topic%20Namespace%20and%20State%20ManagementV2.2-with%20appendix%20B%20format%20-%20Eclipse.pdf) have it as STATE/<PRIMARY ID> and that was also what Cirius-link MQTT Engine used to publish.

I need to modify the code to support both versions of STATE managenment, and I also need to go though the specification to see what else has changed in 3.0.0.

artemz21, 2023-03-01:
No worries.
Please keep in touch on the updates and thanks a lot.

closed 2023-03-06 by thomassorensen2650

thomassorensen2650, 2023-03-06:
I released version 1.4.0 that should address this issue. Let me know if this works as expected

artemz21, 2023-03-06:
Thanks mate, yes it does now. 
Are you planning on implementing the function so the EoN can work in high availability applications so that you can connect to another broker if the comms to the current are lost or Primary Host STATE goes offline as described in section 8.1 of the Sparkplug B spec?

commit b1c8a628 (2023-03-06): Fixed #16 and added Send Birth Immediately func.
commit ecc4b70c (2023-03-06): Merge pull request #17 from thomassorensen2650/spb3.0

Fixed #16 and added Send Birth Immediately func.