thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #21: My Device = Ќ���0  testing/test1 P  testing/test2 P

alexanderberthel, 2023-04-01:
Hi,

when using the "mqtt sparkplug device" node with the Simple Device example I get the above Message in the MQTT Explorer. 
What am I doing wrong? 
NR V 3.0.2
node-red-contrib-mqtt-sparkplug-plus V 1.4.0

all running in an Ubuntu 22 VM

Many Thanks

thomassorensen2650, 2023-04-01:
Sparkplug messages are send as a binary payload.

alexanderberthel, 2023-04-02:
Hello,
Thank you very much for the quick reply. I am quite new to Node Red. 
What is necessary to transfer the messages in a format so that it is readable for people in the MQTT Explorer. Can I set this in the MQTT Explorer or do I have to send the data differently?  I can also ask someone I know. What are the keywords I have to ask to find a solution to my problem?
Thanks a lot!

BasSwildens, 2023-04-03:
Try to find a MQTT client with buildin sparkplug decoding. I use [MQTT.fx](https://softblade.de/en/mqtt-fx/)

[image]

alexanderberthel, 2023-04-04:
that works. many thanks.

closed 2023-04-06 by thomassorensen2650