thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #22: bdSeq missing for NDEATH message

nordfonn, 2023-04-18:
It seems like the NDEATH message coming from the node does not include a bdSeq number (whereas NBIRTH does). This is making it hard to relate the NDEATH to the corresponding NBIRTH, as required by the Sparkplug B v 3.0.0 standard:

> • [tck-id-payloads-ndeath-bdseq] The NDEATH message MUST include the same bdSeq number value
that will be used in the associated NBIRTH message

Here's how the messages look at the SCADA end:

`2023-04-18T07:27:55.087722 spBv1.0/[group_name]/NBIRTH/[eon_name]
		{'timestamp': '1681802847025', 'metrics': [{'name': 'Node Control/Rebirth', 'datatype': 11, 'booleanValue': False, 'value': False}, {'name': 'bdSeq', 'datatype': 1, 'intValue': 0, 'value': 0}], 'seq': '0'}`

`2023-04-18T07:28:02.729686 spBv1.0/[group_name]/NDEATH/[eon_name]
		{'timestamp': '1681802846724'}`

thomassorensen2650, 2023-04-20:
I have fixed the issue (and a few other issues related to bdSeq). I will see if I can release then update this weekend or early next week.

thomassorensen2650, 2023-04-23:
fixed in V1.4.1.

closed 2023-04-23 by thomassorensen2650