thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #23: Birth telegram comes always

mjan-pxcd, 2023-04-21:
Hello, 

it seems that a birth message is always sent even if the option "bith message immediately" is unchecked.

Is this the way it is supposed to be?

Output Birth Topic by first connection:
```
{
  "timestamp" : 1682065348698,
  "metrics" : [ ],
  "seq" : 0
}
```

It would be optimal if no bith massage comes when the option is unchecked, so that no "wrong" birth message is on the mqtt broker to start with.

thomassorensen2650, 2023-04-22:
What is the topic on the birth message NBIRTH or DBIRTH? Also, did you remove the metrics? or do you have a device with 0 metrics?

mjan-pxcd, 2023-04-24:
hi, 

its the birth topic, 
one device with no metrics, send immediately false:
```
{
  "timestamp" : 1682331804692,
  "metrics" : [ ],
  "seq" : 1
}
```
device with metrics. immediately false
--> no birth topic:

device with metrics. immediately true:
```
{
  "timestamp" : 1682332052978,
  "metrics" : [ {
    "name" : "test",
    "dataType" : "Int32"
  }, {
    "name" : "test2",
    "dataType" : "Double"
  } ],
  "seq" : 1
}
```

mjan-pxcd, 2023-04-26:
addendum: 
the zero only occurs when i manually trigger dbirth, 
if I let dbirth write to the device automatically after e.g. 0,1s everything works correctly.

thomassorensen2650, 2023-04-28:
So the problem you are seeing is that a Device with 0 metrics will send birth immediately? 

What difference does the 0.1s make?

closed 2023-08-26 by thomassorensen2650