thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #28: Date as a String

jfaigan, 2023-08-28:
Hello,

I am trying to pass a date value as a String.  The metric is configured thusly:
[image]

I am receiving the following error:
[image]

I understand that the value is a valid date, however, I want this value to be passed as a string to the downstream device.

This error is generated when using the MQTT Sparkplug Out node for a DCMD.

Is there anything I can do to avoid this?  Is there any escape characters that I can use to force the value through as a string?

Thanks in advance.

thomassorensen2650, 2023-09-01:
That definitely sounds like a bug. Would it be possible for you to create a small flow that shows the issue and attach to this issue.

jfaigan, 2023-09-01:
Hi Thomas,

My apologies.  After trying to set up the sample flow I realized that it is just the data presenting itself in the debug as a string when if fact the source data is indeed a date.  I changed my source data to be a string and I am no longer having the issue.

Thank you for reaching out.

This ticket can be closed.

Regards,
Joel

closed 2023-09-09 by thomassorensen2650