thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #32: Inconsistent data when using definition only at start

archerixx, 2023-11-01:
Hi,

I implemented solution, where I defined "definition" programmaticaly at first message together with metrics, then just continued using metrics after that. (like it was mentioned in documentation)
It happens that some of data get lost in transition when sending it like that.

I then tried using definition in each message and with it, data was consistend, no transition problems.

Is there any chance to look into this ?

thomassorensen2650, 2023-11-11:
Hi,

Sorry, need a bit more information. what is the issue you are having.. What is the destination for the data? and what behavior do you see that you don't believe is normal?

archerixx, 2023-11-21:
Issue was that data on SCADA platform (Ignition) was not showing on every data interval. I figured that it was connection issue.

Strange part was that whenever I sent data with "definition" attached to Sparkplug metadata, it would appear on SCADA. Without it, there would be some gaps between data intervals. (I am sending data every 20 seconds, and sometimes data would refresh only every 40 seconds).
Could be that some parts of metrics got lost in transition and Ignition could not parse things correctly.

closed 2023-11-21 by archerixx