thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #35: Not buffering when the Node lost the connection to the broker

alex-controlx, 2023-12-19:
Is any particular reason for "we dont want to publish anything if we are not connected" at
https://github.com/thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus/blob/f9b2d52a1508d658dab4c0ff79273a4c8fc4cc88/mqtt-sparkplug-plus.js#L292

is it a part of the specification?

thomassorensen2650, 2023-12-20:
Hi Alex, 

Thats a really good question, we can't send data before NBIRTH is send -  But... we should save DDATA to the buffer queue, so that it will be send as soon as the connection gets back online.  

You are more than welcome to help create a PR for this, it's just important to also write a unit test to prove problem with current code.

Otherwise I'll see if I can fix it early next year.

alex-controlx, 2023-12-20:
We'll look into this and will create a PR. It needs major restructuring to add unit tests. We'll see what we can do.

Also we found the persistence to the file system using `context` doesn't work as intended. We'll fix it up and create a PR also.

thomassorensen2650, 2023-12-23:
That would be awesome!  The unit test does not need to be anything fancy at all, and it does not need to cover every edge case. Maybe something like like this:

1. Send a few DData
2. Wait 1 second.
3. Connect to MQTT Broker
4. Verify that Data is send after the Birth (or metric data is merged into the birth metrics values)

thomassorensen2650, 2024-02-23:
This has been fixed... will be included next time a version if released.

closed 2024-02-23 by thomassorensen2650