thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #53: Support for Uint Data Types

314PITester, 2024-03-15:
The Node works really well, so I wanted to say thank you for that, but the SparkPlug B protocol supports Uint Data Types.  Any chance you guys could build in support for these?  It would be greatly appreciated.

Thanks,

314PITester, 2024-03-15:
Awesome.  Than you.  I had sent you a message then noticed I could just open a request here so if you see that message over on Node-Red forums you can ignore it.  

When do you think you'll have another release?  Just curious on a few things I'm working on and trying to get a plan around my deployment.

thomassorensen2650, 2024-03-18:
This has been added in V2.1.1.

closed 2024-03-18 by thomassorensen2650

314PITester, 2024-03-20:
This is no longer failing when sending Uint data types.  Thank you for updating this so quickly.