thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #58: npm and node dependencies

Bas-ProRail, 2024-03-19:
Hi, when running sparkplug node installation thru 'manager package' on a fresh nodered docker-compose container with node-red v3.1.7 or v3.1.3 there is an issue with node versions. The last sparkplug package 2.1.1 doesn't have this new node v18 requirements, is that correct?

`2024-03-19T12:04:03.702Z Install : node-red-contrib-mqtt-sparkplug-plus 2.1.2

2024-03-19T12:04:04.967Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-mqtt-sparkplug-plus@2.1.2
2024-03-19T12:04:08.027Z [err] npm
2024-03-19T12:04:08.031Z [err]  WARN
2024-03-19T12:04:08.032Z [err]  config production Use `--omit=dev` instead.
2024-03-19T12:04:17.678Z [err] npm
2024-03-19T12:04:17.679Z [err]  
2024-03-19T12:04:17.680Z [err] ERR! code EBADENGINE
2024-03-19T12:04:17.692Z [err] npm ERR! 
2024-03-19T12:04:17.692Z [err] engine Unsupported engine
2024-03-19T12:04:17.692Z [err] npm ERR! engine
2024-03-19T12:04:17.693Z [err]  Not compatible with your version of node/npm: node-red-contrib-mqtt-sparkplug-plus@2.1.2
2024-03-19T12:04:17.693Z [err] npm 
2024-03-19T12:04:17.694Z [err] ERR! notsup Not compatible with your version of node/npm: node-red-contrib-mqtt-sparkplug-plus@2.1.2
2024-03-19T12:04:17.694Z [err] npm ERR! notsup Required: {"node":">=18.0.0"}
2024-03-19T12:04:17.694Z [err] npm
2024-03-19T12:04:17.694Z [err]  ERR! notsup Actual:   {"npm":"8.19.4","node":"v16.20.2"}
2024-03-19T12:04:17.707Z [err] 
2024-03-19T12:04:17.708Z [err] npm ERR! A complete log of this run can be found in:
2024-03-19T12:04:17.708Z [err] npm ERR!
2024-03-19T12:04:17.709Z [err]      /data/.npm/_logs/2024-03-19T12_04_07_786Z-debug-0.log
2024-03-19T12:04:17.762Z rc=1`

thomassorensen2650, 2024-03-19:
Hi

were you able to install V2.1.1 on node v16? I updated the node dependency version because the MQTT client library requires V18.

thomassorensen2650, 2024-03-20:
I released a new version where i put the requirement back to 12.0.0. I'm curious if you able to install that version.

Bas-ProRail, 2024-03-20:
That does the trick, there is no error anymore on installing and deploying. I found this list on [https://nodered.org/docs/faq/node-versions](url), thank you for reversing the last node version change.

thomassorensen2650, 2024-03-25:
Cool. I can see that I was a bit too aggressive when I upgraded the dependencies for version 2.0.0. I will go through the dependencies and downgrade everything that is not compliment with node 14.

thomassorensen2650, 2024-03-25:
I have updated the package to loosen the requirement for node version. I did a quick test, and I can now install on node-red 2.0.0 with Node 14.

closed 2024-03-25 by thomassorensen2650