thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #62: NodeJS version compatibility

archerixx, 2024-05-16:
Hi,

Is it possible to make version of this module that is compatible with NodeJS version v12.22.12 (which is hard stuck on our Siemens PLC), either making version of this module here or me making some Fork out of it ?

Thanks

thomassorensen2650, 2024-05-19:
Did you already look into what (if any) dependencies needs to be downgraded in package.json?

archerixx, 2024-06-04:
Sorry for late response. These are packages that are unsupported for NodeJS v12.22.12:

package: 'https-proxy-agent@7.0.4',
package: 'mocha@10.4.0',
package: 'node-red-node-test-helper@0.3.4',
package: 'agent-base@7.1.1',
package: 'supertest@7.0.0',
package: 'superagent@9.0.2'

I used packages from version 1.4.1 and built your 2.1.7 release version. Seems that all fine with node-red and package, but I will test throughout week and let you know.

archerixx, 2024-06-23:
Lib is working fine with previously mentioned packages.

closed 2024-06-23 by archerixx