thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #63: Missing important "clean session" checkbox on mqtt broker settings

JSmits96, 2024-06-05:
Cannot find a way to enabled/disable clean session for mqtt broker configuration.

thomassorensen2650, 2024-06-05:
This checkbox was actually removed to clean up the UI. Per the Sparkplug B specification connections should always use clean session. Is there any reason why you would like to not use a clean session?

JSmits96, 2024-06-05:
That seems strange that it is not in the specification, I'm not aware of that. 
for Spb mqtt clients I would need my session to be persisted to make sure I can recover from data connectivity issues without data loss. I tested locally and I lose data when my client reconnects to the broker.
Is there another way to overcome this?

thomassorensen2650, 2024-06-05:
Are you using the Sparkplug Device node? Do you have the "Store Forward when not connected" option enabled on that Sparkplug Device?

JSmits96, 2024-06-07:
I'm using the 'mqtt sparkplug in' node, not the EoN device node. 
For this node I do see added value in having the session. I agree that for EoN nodes, that are mainly metrics generators, there is no need as it is also specified in the specification.

As a backup solution I could switch to standard mqtt node, with session option, followed by a proto decoder but I would highly prefer to stick to your solution here for the 'mqtt in' nodes as it works perfectly fine.