thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #64: Alias for metrics breaks NBirth/NDeath certificates

archerixx, 2024-06-23:
Hi,

I am using Ignition as SCADA and so far I was using options to Disable BD Seq validation. I checked in your release logs that bdSeq should be working fine here, so I tried testing it out on my project.

I found out that when I am using Aliases, MQTT and SCADA do not recognize that Node is dead.
But when I disable Aliases for Metris option, everything works fine.

Would you have any idea why this would be case? Is it possible to have it working with Aliases here ?

**Here are logs of NBIRTH payload (with Aliases):**
{"timestamp":"2024-06-23T21:22:01.375Z","metrics":[{"value":false,"type":"Boolean","name":"Node Control/Rebirth","alias":{"low":2,"high":0,"unsigned":true}},{"value":{"low":0,"high":0,"unsigned":true},"type":"UInt64","name":"bdSeq","alias":{"low":1,"high":0,"unsigned":true}}],"seq":0}
**And NDEATH payload (with Aliases):**
{"timestamp":"2024-06-23T21:22:01.356Z","metrics":[{"value":{"low":0,"high":0,"unsigned":true},"type":"UInt64","alias":{"low":1,"high":0,"unsigned":true}}],"seq":0}

**Here are logs of NBIRTH payload:**
{"timestamp":"2024-06-23T21:25:43.359Z","metrics":[{"value":false,"type":"Boolean","name":"Node Control/Rebirth"},{"value":{"low":0,"high":0,"unsigned":true},"type":"UInt64","name":"bdSeq"}],"seq":0}
**And NDEATH payload:**
{"timestamp":"2024-06-23T21:25:43.341Z","metrics":[{"value":{"low":0,"high":0,"unsigned":true},"type":"UInt64","name":"bdSeq"}],"seq":0}

Thanks in advance for looking into this

thomassorensen2650, 2024-06-24:
Strange.. Do you get any warnings/error messages in Ignition?

thomassorensen2650, 2024-06-24:
One issue I can see is that bdSeq is suppose to be INT64 and not UNIT64, but I don't think that is causing the issue you see.

From the example payload you send, it looks like the bdSeq in the NDEATH message matches the bdSeq in the BIRTH seq, so it should work.

From the Specification:
_Imediately upon reception of an Edge Node Death Certificate (NDEATH message) with a bdSeq number that matches the preceding bdSeq number in the NBIRTH, any Host Application subscribed to this Edge Node should set the data quality of all metrics to STALE and should note the timestamp when the NDEATH message was received._

archerixx, 2024-06-24:
Last time I did it, there were not any warnings or errors in Ignition, but I will try running it tonight and post logs here

archerixx, 2024-06-25:
My guess is that in NDEATH alias is not recognized since "name" is not being sent alongside it. In NBIRTH "name" is sent alongside alias.
I understand that is how it should work with aliases, but maybe in these Certificate commands it does not work similarly.

thomassorensen2650, 2024-07-08:
I think I found the issue (bdseq was increased on rebirth). I'll see if I can release a new version with the fix soon.

thomassorensen2650, 2024-07-18:
This issue should be fixed in version 2.1.9 (just released).

archerixx, 2024-07-19:
That did it. Thanks for updates

archerixx, 2024-07-31:
So,

thomassorensen2650, 2024-08-01:
Interesting. what do you mean "skip" alias assign to NBIRTH? is this still related to bdSeq?

thomassorensen2650, 2024-08-01:
PR for the other issue looks good to me, I will get that merged in.. If there is another issue, please create a new PR. That makes it a lot easier for me to track. 

And if we need to skip alias assign, then we need to consult with the Sparkplug B specification to see if thats ok per the specification.

archerixx, 2024-08-01:
I will create PR for this issue then separately and try to explain in PR what I meant by it

archerixx, 2024-08-02:
@thomassorensen2650  will you be able to check and merge PRs over weekend ?

thomassorensen2650, 2024-08-31:
Can you check if it works with v2.1.10 I just released? This version will not add alias to bdSeq..

archerixx, 2024-09-03:
That solved the issue, thanks.

closed 2024-09-03 by archerixx

thomassorensen2650, 2024-09-19:
There are a bunch of places where we modify the message. 

I think we should deep clone the message (where you added the shallow copy in the last PR.  Mabye RED.util.cloneMessage(msg); can be used for this?