thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #65: Bug in code for sending Historical data if host client is off

ghocev1, 2024-07-01:
[Testing history data.docx](https://github.com/user-attachments/files/16049146/Testing.history.data.docx)

Hello Thomass,

I collaborated with Wes from Cirrus Link and Inductive Automation (Ignition SCADA) to investigate an issue related to historical data not functioning correctly.

We observed that when the MQTT Distribution (MQTT Broker) module was toggled OFF and ON, historical data worked as expected. **However, when we switched the MQTT Engine (host client), historical data stopped functioning**. Upon re-establishing the connection, the Node Red Edge Node sent a DBIRTH message, which is invalid because it should send an NBIRTH first. Consequently, historical metrics were transmitted, but due to improper initialization of the Node Red client, they were effectively lost.

In this scenario, the Node Red code is not Sparkplug compliant, especially when switching the MQTT Engine OFF and back ON. As a result, historical data does not appear.

I’ve attached Cirrus Link’s response for your reference.

Thank you for your answer.

thomassorensen2650, 2024-07-03:
Thanks for doing all the investigation - this is awesome! ... I'll see if I can get a fix implemented as soon as possible.

thomassorensen2650, 2024-07-05:
I just released V2.1.8 with the fix. Hopefully this will solve your issue

ghocev1, 2024-07-08:
Thank you. This fix is solvig this issue.

closed 2024-07-08 by thomassorensen2650