thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #72: The datatype SHOULD NOT be included with metric definitions in NDATA, NCMD, DDATA, and DCMD messages.

NklsP, 2024-08-09:
Hi @thomassorensen2650,

Acording to the specification, the datatyps should not be included with metric definitions in NDATA, NCMD, DDATA, and DCMD messages.

When sending the following payload to the mqtt-sparkplug-out node the following error appears.

```js
msg.payload = {
      "metrics": [
          {
              "name": "Control",
              "value": [
                  1
              ]
          }
      ]
  }
```

> "Unable to encode  message. the error is : Unable to encode message, all metrics must have a 'type' Attribute"

Error should only appear when publishing a NBIRTH and DBIRTH.

thomassorensen2650, 2024-08-10:
Aha, I can see that dataType is now optional in V3.0.0 of the Sparkplug Specification, I will see if I can remove the warning for Sparkplug in and Sparkplug out.

I will keep adding the data types in the Sparkplug Device node to keep the nodes compatible with V2.0.0 of the Sparkplug Specification as well.

For the type vs. dataType: This is how it's implemented in the official sparkplug-payload package. I assume that they did not change it due to backward compatibility.  Im a bit hesitant to change this, as it could break a lot if existing flows. This will be change for a version 3.0.0 of the nodes.

NklsP, 2024-08-12:
Ok, removing the warnign should be enough. 

Maybe a checkbox to disble this behaviour until v3.0.0 of the node?

thomassorensen2650, 2024-08-25:
I thought that this would be an easy fix, but it's a bit more complicated than just removing the error message - I actually need the datatype in order to protobuf encode the message.

Here is a url to the "official" sparkplug-payload encoder function that shows the issue.

https://github.com/eclipse/tahu/blob/5736e404889d4b95910613040a99ba79589ffb13/javascript/core/sparkplug-payload/lib/sparkplugbpayload.ts#L101

I could do some custom logic that removes the type for metrics after the value has been encoded, if message type is DDATA, but as the user would still need to add the datatype when sending from node-red AND the message is already compliant the way its done today where we send the type (its a should requirement and not a must requirement). then it seems like a lot of work with no real benefit. Also keeping it the way it is today makes the nodes both sparkplug V2 and V3 compliant at the same time.

thomassorensen2650, 2024-08-28:
I will close this ticket, as this would be a nice to have, that is very low proiority. I need the datatype to encode the value correctly, so the only thing I could do would be to remove one or two bytes for the datatype from the binary payload

closed 2024-08-28 by thomassorensen2650