thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #74: Dynamic metric definitions - can msg.definition be an array ?

industrialinternet, 2024-08-27:
Thomas thanks for your continued good work on Sparkplug-plus.
For my use case I have 100s of metrics each with at least two properties or more these are stored as JSON
So question for me is can msg.definition be an array?
Or does each metric need to be sent separately when using Dynamic metric definitions?

Thanks

```
msg ={
  "definition" : [
    {
      "outdoor_pallet_area_1/externalTemperature": {
        "dataType": "Float",
        "properties": {
          "engUnits": {
            "type": "String",
            "value": "C"
          },
          "devEUI": {
            "type": "String",
            "value": "A81758FFFE0375FF"
          }
        }
      }
    },
    {
      "outdoor_pallet_area_1/humidity": {
        "dataType": "UInt16",
        "properties": {
          "engUnits": {
            "type": "String",
            "value": "%RH"
          },
          "devEUI": {
            "type": "String",
            "value": "A81758FFFE0375FF"
          }
        }
      }
    }
  ]
}
```

thomassorensen2650, 2024-08-27:
Im not sure that I understand the issue. Definition is an object that can have multiple attributes.   Why do you want it to be an array? 

```
msg = {
    "definition": {
        "outdoor_pallet_area_1/externalTemperature": {
            "dataType": "Int32",
            "properties": {
                "engUnits": {
                    "type": "String",
                    "value": "inHg"
                }
            }
            },
            "outdoor_pallet_area_1/humidity": {
                "dataType": "UInt16",
                "properties": {
                    "engUnits": {
                        "type": "String",
                        "value": "%RH"
                    }
                }
            }
    }
}
```

industrialinternet, 2024-08-27:
Hi it would just make my processing a little easier but no worries.
BTW have you tested with Ignition SCADA? 
I looked at sparkplug validation tool that looks a little too much so I'm going to use Ignition.
Again thanks for your work on this node much apricated.

thomassorensen2650, 2024-08-28:
If you ever get that Sparkplug validation tool to work let me know, that would be super interesting.. Yes, multiple people use the nodes with Ignition.

I will close this issue, as I dont see any action.

closed 2024-08-28 by thomassorensen2650