thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #76: EoN Node Stays in "Connected" Status When Broker Connection is Lost

carloslockward, 2024-09-03:
## Issue: EoN Node Stays in "Connected" Status When Broker Connection is Lost

### Description
When an EoN is configured with a primary host application, if the physical connection to the broker is lost (e.g., broker's ethernet cable unplugged), the EoN does not receive a death message and therefore remains in the "connected" status.

### Steps to Reproduce
1. Configure an EoN node with a primary host application. (In my particular case, the Broker and primary host app are in the same device)
2. Ensure the EoN node is connected to the broker and in "connected" status.
3. Physically disconnect the broker's ethernet cable to simulate a lost connection.
4. Observe the status of the EoN node.

### Expected Behavior
The EoN node should transition to a "disconnected" or appropriate error status upon losing the physical connection to the broker.

### Actual Behavior
The EoN node remains in "connected" status despite the loss of physical connection to the broker.

archerixx, 2024-09-09:
@carloslockward in my case, it always takes around 2/3 minutes for Node to go in offline state.
Did you try subscribing to NDEATH topic to check if your broker ever sends death certificate ?

carloslockward, 2024-09-09:
> @carloslockward in my case, it always takes around 2/3 minutes for Node to go in offline state.
> Did you try subscribing to NDEATH topic to check if your broker ever sends death certificate ?

It would be impossible for the Broker to send the NDEATH if you physically disconnect the Ethernet cable.

archerixx, 2024-09-09:
Your MQTT broker (server), that your Nodes communicate with, receives from them LWT (last will message) which contains DEATH certificate which is later sent to your Main Application (some SCADA system or similar) in case your Node device gets disconnected to let you know that Node is no longer Online.
It would be best to first look into how things work here on these links:
http://www.steves-internet-guide.com/sparkplug-payloads-and-messages/
https://www.hivemq.com/blog/mqtt-sparkplug-session-management-iiot/

Best of luck

archerixx, 2024-09-10:
Oh, you meant that text under sparkplug nodes ?
I tried reproducing, but I got "disconnected" after few seconds.

Do you have anything in node-red console like these logs bellow ?
```
 [info] [mqtt-sparkplug-broker:Ignition] Connected to broker: mqtt://172.x.x.x:1883
 [info] [mqtt-sparkplug-broker:Ignition] Disconnected from broker: mqtt://172.x.x.x:1883
```

carloslockward, 2024-09-10:
> Oh, you meant that text under sparkplug nodes ? I tried reproducing, but I got "disconnected" after few seconds.
>
> Do you have anything in node-red console like these logs bellow ?
>
> ```
>  [info] [...]

I need to reproduce it again and look at the logs. Only thing I observed was that the node remained `connected` even after several minutes. 

Could you give me more details on your test setup for this?

archerixx, 2024-09-10:
I did notice that if sparkplug node is not triggered, it will not go into disconnect state. But as far as I know this service does not check if server is alive or not constantly, only upon request.

thomassorensen2650, 2024-09-19:
Do you see the same issue with the MQTT nodes that er build-in to node-red?

thomassorensen2650, 2024-09-19:
Also, can. you try to time how long time it will take before it goes into "disconnected" state? I think the default is 2 minutes

ghocev1, 2025-01-24:
@thomassorensen2650 If **I use only one MQTT Sparkplug device in Node-RED and all data are written in the same metrics, the "connected/disconnected status" works**. However, if I use multiple MQTT Sparkplug devices, the "connected status" does not work properly. For one MQTT Sparkplug device, the "connected status" works, but for other MQTT Sparkplug devices, the EoN does not receive a death message and therefore remains in the "connected" state.

My main issue is that if I have a lot of tags in the metrics, it becomes cluttered. It would be nice if the connected status on Ignition SCADA would work even if I use multiple MQTT Sparkplug devices or if I could use more separate metrics in the same MQTT Sparkplug device to create a structure and separate tags.

Thank you for your answer.

ghocev1, 2025-01-31:
I have a similar issue. I have been waiting for 15 minutes, and I still haven’t gotten all the tags disconnected.

ghocev1, 2025-01-31:
@thomassorensen2650   Currently, I am testing the operation of an MQTT Sparkplug B Device with a demo Ignition SCADA and a free MQTT Broker Serverless HiveMQ. I have Node-RED installed on a Raspberry Pi 4, which is connected via Ethernet to a 4G router and a Siemens S7-1200 PLC.

In Node-RED, I have linked all tags to a single MQTT Sparkplug B Device, hoping that the connect/disconnect status would work flawlessly. I have enabled store and forward for all tags. The connection to the MQTT broker is constant, but the connection to SCADA disconnects every two hours.

When the MQTT Sparkplug B Device loses connection with SCADA, at least some tags immediately get a disconnected status, but most of the time, not all tags do, even after 15 minutes. When I restart Ignition SCADA, sometimes not all tags get a connected status, only those whose values have changed.

If I make any changes in the Node-RED program and confirm them with Deploy, all tags immediately get a connected status. During the test, I mostly changed the analog value on the Siemens controller, while I rarely changed other tags.

It would be really nice if the connect/disconnect status worked without issues.

ghocev1, 2025-02-03:
[image]

The attached image shows the error when tags do not get the connected status after Ignition SCADA reconnects to the MQTT broker.
I did get all historical data only all tags do not get connected status back until I do not restart Node red or MQTT Sparkplug B device sends new value for that tag.