thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #85: Automatic NBIRTH with DataSet definition

ale075, 2024-10-17:
Dataset type is supported, but isn't it possible to define the columns and send NBIRTH automatically? 

It's possible to use Device module to send DBIRTH and DDATA? sequence number is generated internally?

thomassorensen2650, 2024-10-18:
So you want to send a empty dataset with column definitions on birth? 

1. Make sure that "Send Birth Immediately" option is disabled on the device node.
2. Create a inject node with a msg.payload with the default values you want. and enable "Inject once after" checkmark. 

Now the device will birth with what ever you configured in that inject node. 

Not sure I understand the last question (seq is managed internally by the node).

ghocev1, 2025-01-24:
Is it possible to read what is the currend seq from Device module or the only way is to subscribe to server and read seq from the MQTT server.

ghocev1, 2025-01-24:
> Dataset type is supported, but isn't it possible to define the columns and send NBIRTH automatically?
> 
> It's possible to use Device module to send DBIRTH and DDATA? sequence number is generated internally?

I you are sending data to Device module everything is done automaticaly by this module. If you want to send data manually with **out node** you need to know seq number ( for now I am reading this number from messeges on MQTT server). With each new message you need increase numer by one. ( seq=count+1) count needs to be smaller than 255. If count is bigger you need to set count to 0 and start again. this count needs to be done by you in node red. All data types are working if you implement sparkplug B specification correctly.