thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #89: Rebirth does not work properly

ghocev1, 2025-02-12:
Hello.

Sometimes it happens that the connect/disconnect status does not work properly. When EON loses the connection with SCADA, it starts storing data. I have all the tags linked to one Sparkplug B device with the "Store Forward when not connected" feature enabled. When the connection with Ignition SCADA is reestablished, occasionally some tags get the status "Good" while others get "Bad_Stale". **When I trigger a Rebirth in Ignition SCADA, Sparkplug B sends the DBIRTH message and it appears on the MQTT broker, but the connect/disconnect status does not refresh in Ignition SCADA. The only thing that always works is to either power cycle the edge PC or press Deploy in Node-RED or when the value of a tag with a Bad_Stale status changes.**
I am using Free HiveMQ Cloud and demo Igniton SCADA for testing.
It is interesting that when I push Deploy button in Node-Red tags always reinitialize its state but it never works when I use Rebirth command on Ignitino SCADA.

[image]

I did find some errors in the log.

[image]

[image]

[image]

This is what I get when I trigger Rebirth control in Ignition SCADA

[image]

ghocev1, 2025-02-18:
[image]

ghocev1, 2025-02-18:
[image]

ghocev1, 2025-02-27:
Hello.
I believe that the problem is in the rebirth procedure. Based on the Ignition log file, the edge node sends DDATA before it gets the BIRTH message, which must include the entire data structure for all metrics for this device.

[image]

### Upon receipt of a Rebirth Request, the Edge Node must do the following.
• [tck-id-operational-behavior-data-commands-rebirth-action-1] When an Edge Node receives a
Rebirth Request, it MUST immediately stop sending DATA messages.
• [tck-id-operational-behavior-data-commands-rebirth-action-2] After an Edge Node stops
sending DATA messages, it MUST send a complete BIRTH sequence including the NBIRTH and
DBIRTH(s) if applicable.
• [tck-id-operational-behavior-data-commands-rebirth-action-3] The NBIRTH MUST include the
same bdSeq metric with the same value it had included in the Will Message of the previous
MQTT CONNECT packet.
◦ Because a new MQTT Session is not being established, there is no reason to update the bdSeq
number
**• After the new BIRTH sequence is published, the Edge Node may continue sending DATA messages.**

thomassorensen2650, 2025-03-07:
Would it be possible for you to investigate some more about what the actual problem is? Maybe have node-red output all topic to see if it actually sends DDATA before birth are send.

ghocev1, 2025-03-11:
Hello. Thank you for the response. I will try to find the error. For now, I suspect the error might be in the wrong sequence number, but I can't verify it at the moment because I have the Edge PC at another location. Free HivemMQ cloud allows connection from only one computer. If I subscribe to MQTT sparkplug b messages from another computer, it starts blocking the connection.

I will  remove the Edge PC and bring it to the workshop within the next two weeks to diagnose the error.

If the tags do not appear when SCADA gets the ONLINE status, NCMD "Node Control/Rebirth" triggered from SCADA never helps. Likewise, if I restart the computer with Ignition SCADA installed, the tags often do not appear in SCADA. The NCMD "Node Control/Rebirth" command from SCADA does not help either.

When the PLC tag values change, the data appears in SCADA.

Temporarily, I solved this problem by adding the node **node-red-contrib-resend-last-msg** to each tag. When this node receives a value, it sends it to the output and stores the value until it receives a new one. I connected the Resend Last value node to an external trigger Reset. When I manually trigger the reset from SCADA, the values of all tags are sent again to the Sparkplug B device. This way, the tags always appear in SCADA. The PLC tag values are set to RBE, so the value is sent only when it is changed.

ghocev1, 2025-03-11:
If it will help with diagnostics, I am attaching the Node-RED flow I created while learning about MQTT Sparkplug B communication. When I manually trigger the "function Rebirth" here, NDATA and DDATA always appear in SCADA. I made this with my limited programming knowledge since I am not a programmer but an electrical engineer. Maybe this flow can help you resolve the connectivity issue between the Sparkplug B device and SCADA.

I also commented in thread #76 EoN Node Stays in "Connected" Status When Broker Connection is Lost with the hope that my description of the issue might assist in resolving the problem.

[Sparkplug_B test_flow.json](https://github.com/user-attachments/files/19178158/Sparkplug_B.test_flow.json)

I created this Node-RED flow based on the example of Sparkplug B messages from the documentation below, mostly from page 61 onwards. 

[Sparkplug Topic Namespace and State ManagementV2.2-with appendix B format - Eclipse.pdf](https://github.com/user-attachments/files/19178316/Sparkplug.Topic.Namespace.and.State.ManagementV2.2-with.appendix.B.format.-.Eclipse.pdf)

thomassorensen2650, 2025-03-11:
What is the attached flow use for? It looks like you have tried to crate your own EoN node? 

Just to make sure that I undestand this connectedly; the rebirth issue you are experience is with the "MQTT Sparkplug Device" node - right?

ghocev1, 2025-03-11:
The attached flow is used for my understanding of how Sparkplug B is working. The rebirth issue  is with the "MQTT Sparkplug Device" node.

ghocev1, 2025-03-11:
I tested the difference between the MQTT Transmission module and the MQTT Sparkplug B Plus Device in the implementation of 'Node Control/Rebirth'. The sequence for Rebirth using the Transmission module is: 

[image]

1.) Node Control/Rebirth,
2.) NBIRTH
3.) DBIRTH

The sequence for Rebirth with **MQTT Sparkplug Device" node** is: 

[image]

1.) Node Control/Rebirth,

 **2.) NDEATH**
 3.) NBIRTH
 4.) DBIRTH

It could be that the NDEATH, which the MQTT Transmission module from Cirrus Link does not use, is causing the error.

ghocev1, 2025-03-11:
The next difference is in the sequence number

[image]

[image]

[image]

You can check in my flow for the function NDEATH. I believe that I created the NDEATH message correctly according to the Sparkplug B specification.

ghocev1, 2025-03-31:
Hello @thomassorensen2650 

Can you remove the NDEATH message from the Rebirth sequence? If the NDEATH message is used anywhere else, please remove the sequence number as well. I think that this would solve the problem.

thomassorensen2650, 2025-05-16:
Sorry, I haven't had time to look at this. The NDEATH was required in V2.2 of the specification, I need to read through the V3 specification one more time, to make sure it's handled correctly.

It would be really great, if we could get a TCK setup to to a fully check that we conform to all the requirements, but I haven't had time to do it.

https://www.hivemq.com/blog/starter-guide-sparkplug-3-0-certification-technology-compatibility-kit-tck-industry-40/

ghocev1, 2025-05-16:
Thank you for your response.

I went through the Sparkplug B specification several times, and I did not find any indication that NDEATH is required for rebirth. Then, I tested rebirth in the MQTT Transmission module (which has already been TCK checked), and there was no NDEATH message. Additionally, **the NDEATH message MUST NOT include a sequence number.**

[image]

[image]

[image]

Thank you again for your response .I really appreciate it.

silviozambon-toffoli, 2026-05-13:
I can confirm the issue is still present to this day. How can I apply the hotfix waiting for the patch to be release? Thank you