thomassorensen2650/node-red-contrib-mqtt-sparkplug-plus #92: Rebirth command is broken in V2.1.11

Megatol75, 2025-06-04:
Looks like Rebirth command is broken in v2.1.11 but works fine in v2.1.10

How to reproduce:
- I have [MQTT sparkplug device] Node with 4 metrics configured
- I have [Inject Send Metrics] Node wich will send these 4 metrics
- I have [Inject DBIRTH] Node with will send rebirth comand: 
msg = {
    "command" : {
        "device" : {
            "rebirth" : true
        }
    }   
};

1. Click on [Inject Send Metrics] - you will get GOOD DBIRTH message for 4 metrics with aliases from 1 to 4
2. Click second time on [Inject Send Metrics] - you will get GOOD DDATA message for 4 metrics with aliases from 1 to 4
3. Click on [Inject DBIRTH] - you will get BAD DBIRTH message (it is for all 4 metrics but all names are missing and all aliases equal to 5), I expect it to be same as step 1

ghocev1, 2025-06-11:
Import this node and connect with Sparkplug B Device. It should work

[
    {
        "id": "a82eb3893a36b69f",
        "type": "inject",
        "z": "b8cc857420752679",
        "name": "Send rebirth command",
        "props": [
            {
                "p": "command",
                "v": "{\"device\":{\"rebirth\":true}}",
                "vt": "json"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "x": 580,
        "y": 20,
        "wires": [
            [
                "45762de3465f5444"
            ]
        ]
    }
]

Megatol75, 2025-06-11:
Yes, I have exactly same node already and rebirth command works in v2.1.10 and below and doesn't work in v2.1.11

ghocev1, 2025-06-12:
I belive it does not work because in rebirth sequence there is NDEATH message with seq number. 

[image]

[image]